tuanqing/mknop #15: 希望大佬能给点写入emmc的指点

nnssz, 2020-03-31:
参考了您的：[N1盒子] 【2019-09-05更新】贝壳云 / N1一键制作OpenWrt镜像脚本 <=> 定制你的专属固件 https://www.right.com.cn/forum/thread-777305-1-1.html  16楼回复：
用您的脚本生成的openwrt启动N1，此时一切运行良好。
然后，shell中删除/mnt/mmcblk1p2/分区下文件，并写入/mnt/sda2/所有文件到/mnt/mmcblk1p2/分区。
之后重新启动，N1无法联网，可以用hdmi连接看到N1启动过程。

是不是跟N1先前安装的armbian版本有关呢，需要事先安装什么armbian版本才行，希望能指点一二，感谢。

tuanqing, 2020-03-31:
可以尝试使用armbian官方的脚本，但是肯定缺少对应的可执行文件，需要自行对照缺少哪些可执行文件，在openwrt上对应编译或者从armbian中提取

hcym, 2020-03-31:
现在的f大大底包越来越复杂，我还是用博主去年九月份xq7的备份，感觉没什么差别，反应速度更快🤣

tuanqing, 2020-03-31:
> 参考了您的：[N1盒子] 【2019-09-05更新】贝壳云 / N1一键制作OpenWrt镜像脚本 <=> 定制你的专属固件 https://www.right.com.cn/forum/thread-777305-1-1.html 16楼回复：
> 用您的脚本生成的openwrt启动N1，此时一切运行良好。 [...]

tuanqing, 2020-03-31:
> 参考了您的：[N1盒子] 【2019-09-05更新】贝壳云 / N1一键制作OpenWrt镜像脚本 <=> 定制你的专属固件 https://www.right.com.cn/forum/thread-777305-1-1.html 16楼回复：
> 用您的脚本生成的openwrt启动N1，此时一切运行良好。 [...]

这个只拷贝了rootfs分区，boot分区没有拷贝过去，版本不一样的话会有问题

nnssz, 2020-03-31:
就是说，N1原本要刷入emmc的armbian底包要和您mknop脚本中的一致才行，对吧？
那您现在用的是armbian底包具体的标本号能否发一下，我去Google下载下来，麻烦了❤

nnssz, 2020-03-31:
> 现在的f大大底包越来越复杂，我还是用博主去年九月份xq7的备份，感觉没什么差别，反应速度更快

hcym大佬，感谢您的提示。您是按照MAX橙子大佬的教程，一步一步手动做的吗，还是另外的方法，方便的话请稍微指点一二，🙏

hcym, 2020-04-01:
哦，手动也可以，就是橙子的方法，我说的是用博主去年的旧版自动生成，底包是xq7的，内核4.18，就是openssl比较老，其他的都很好

tuanqing, 2020-04-01:
> 就是说，N1原本要刷入emmc的armbian底包要和您mknop脚本中的一致才行，对吧？
> 那您现在用的是armbian底包具体的标本号能否发一下，我去Google下载下来，麻烦了❤

https://yadi.sk/d/srrtn6kpnsKz2/Linux/ARMBIAN/20.02/20200205 官方最新
你把boot分区也拷过去应该可以吧，如果无法启动，再考虑用新armbian install

nnssz, 2020-04-01:
> 哦，手动也可以，就是橙子的方法，我说的是用博主去年的旧版自动生成，底包是xq7的，内核4.18，就是openssl比较老，其他的都很好

感谢hcym大佬指点，记下了👍

nnssz, 2020-04-01:
> > 就是说，N1原本要刷入emmc的armbian底包要和您mknop脚本中的一致才行，对吧？
> > 那您现在用的是armbian底包具体的标本号能否发一下，我去Google下载下来，麻烦了❤
> 
> https://yadi.sk/d/srrtn6kpnsKz2/Linux/ARMBIAN/20.02/20200205 官方最新
> 你把boot分区也拷过去应该可以吧，如果无法启动，再考虑用新armbian install

感谢tuanqing大佬，yadisk上的20200205armbian已经下载，回头试试，🙏

nnssz, 2020-04-01:
> > 就是说，N1原本要刷入emmc的armbian底包要和您mknop脚本中的一致才行，对吧？
> > 那您现在用的是armbian底包具体的标本号能否发一下，我去Google下载下来，麻烦了❤
> 
> https://yadi.sk/d/srrtn6kpnsKz2/Linux/ARMBIAN/20.02/20200205 官方最新
> 你把boot分区也拷过去应该可以吧，如果无法启动，再考虑用新armbian install

感谢tuanqing大佬，yadi.sk上的20200205 armbian已经下载，回头试试，🙏

nnssz, 2020-04-02:
感谢tuanqing和hcym大佬，底包使用https://yadi.sk/d/srrtn6kpnsKz2/Linux/ARMBIAN/20.02/20200205的Armbian_20.02.0-rc1.036_Arm-64_buster_current_5.5.0-rc6_20200205.img后成功写入N1的emmc，并启动openwrt，科学上网也没问题。

具体过程：
1. windows中用rufus将Armbian_20.02.0-rc1.036_Arm-64_buster_current_5.5.0-rc6_20200205.img写入U盘；
2. U盘有两个分区，其中BOOT分区为fat16（vfat）格式，windows可以识别出，也可以用带usb口的openwrt路由器挂载）。
 修改U盘中BOOT分区的uEnv.txt的配置为：
# aml s9xxx
FDT=**/dtb/amlogic/meson-gxl-s905d-phicomm-n1.dtb**
APPEND=root=LABEL=ROOTFS rootflags=data=writeback rw console=ttyAML0,115200n8 console=tty0 no_console_suspend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0
3. 
N1此时的emmc中应该已经刷入某个armbian系统或webpad的v2.2android系统，
N1断电，
将U盘插入N1挨着hdmi口的usb口，
N1通电，
此时N1应该启动进入U盘中的Armbian_20.02.0-rc1.036_Arm-64_buster_current_5.5.0-rc6_20200205系统。
4. 
输入用户名root，默认密码1234
再次输入密码1234
输入新的至少8位以上的密码，比如：password
输入第二遍password
输入ctrl+c
再次输入用户名root，密码password
进入U盘armbian系统。
5. 输入下面命令将Armbian_20.02.0-rc1.036_Arm-64_buster_current_5.5.0-rc6_20200205底包本身，刷入N1的emmc中：
bash  /root/install-aml.sh
6. 用tuanqing大大这个仓库mknop，处理自己编译的lean大或者lienol大的RasPI 3B+ openwrt系统，得到一个N1的openwrt镜像，比如20.04.02-214329-phicomm-n1-arm64-openwrt-firmware.img文件传输到U盘armbian系统的/root目录下。
7. 按照Max橙子的手动方法（https://www.maxlicheng.com/embedded/36.html），
将N1的emmc的第二分区mmcblk1p2，挂载到一个文件夹/emmc2，删除/emmc2目录下的所有文件，
再将20.04.02-214329-phicomm-n1-arm64-openwrt-firmware.img镜像挂载，其root分区到/media目录下，将/media目录下的所有文件复制到/emmc2目录下。
卸载所有的挂载目录和镜像，poweroff关机。
拔出U盘，再开机，
done！

自己小白，啰里啰唆一大堆，记录下来备忘，也方便其他小白。

好消息是，刚刚看到tuanqing大佬准备发大招，制作方便openwrt写入emmc的脚步了，期待中！

closed 2020-04-03 by nnssz