tuanqing/mknop #16: 不能启动

susanooxyz, 2020-04-03:
用自己编译的openwrt打包的固件不能启动，用openwrt目录中自带的打包生成的固件也是不能启动。
自己编译的openwrt固件Target System选择的是QEMU ARM Virtual Machine,Subtarget是ARMv8
其他都没有改动。U盘没有问题，用其他固件可以正常启动。

tuanqing, 2020-04-03:
> 用自己编译的openwrt打包的固件不能启动，用openwrt目录中自带的打包生成的固件也是不能启动。
> 自己编译的openwrt固件Target System选择的是QEMU ARM Virtual Machine,Subtarget是ARMv8
> 其他都没有改动。U盘没有问题，用其他固件可以正常启动。

是连显示器看的嘛？确实看不到OpenWrt这个画面。不过用网线连接进后台正常

hcym, 2020-04-03:
大佬，请问你这个底包是105的还是f大的？ 105哪个用op的话，切换节点开关防火墙非常非常慢，我用105的编译最简版也慢的不行，f大的没看到源码，但是用f底包就没切换防火墙的问题。不过总的来说，5.0以后的多多少少都有不稳定性，有点看运气

tuanqing, 2020-04-03:
> 大佬，请问你这个底包是105的还是f大的？ 105哪个用op的话，切换节点开关防火墙非常非常慢，我用105的编译最简版也慢的不行，f大的没看到源码，但是用f底包就没切换防火墙的问题。不过总的来说，5.0以后的多多少少都有不稳定性，有点看运气

官方的20.2

hcym, 2020-04-03:
哦，难怪切换很慢，150balbes的记忆
那个用armbian不错，做op真的不合适
不知道什么原因，flippy的没这个问题，但是没有xq7那个稳定

susanooxyz, 2020-04-03:
> > 用自己编译的openwrt打包的固件不能启动，用openwrt目录中自带的打包生成的固件也是不能启动。
> > 自己编译的openwrt固件Target System选择的是QEMU ARM Virtual Machine,Subtarget是ARMv8
> > 其他都没有改动。U盘没有问题，用其他固件可以正常启动。
> 
> 是连显示器看的嘛？确实看不到OpenWrt这个画面。不过用网线连接进后台正常

是连的显示器，启动代码不再跑之后，按回车键不能退出。用网线也连不上n1,感觉是没启动完成。

tuanqing, 2020-04-04:
> > > 用自己编译的openwrt打包的固件不能启动，用openwrt目录中自带的打包生成的固件也是不能启动。
> > > 自己编译的openwrt固件Target System选择的是QEMU ARM Virtual Machine,Subtarget是ARMv8
> > > 其他都没有改动。U盘没有问题，用其他固件可以正常启动。
> > 
> > 
> > 是连显示器看的嘛？确实看不到OpenWrt这个画面。不过用网线连接进后台正常
> 
> 是连的显示器，启动代码不再跑之后，按回车键不能退出。用网线也连不上n1,感觉是没启动完成。

不会，连上之后给你电脑手动指定一个192.168.1.0/24网段的IP就行了

tuanqing, 2020-04-04:
> > > 用自己编译的openwrt打包的固件不能启动，用openwrt目录中自带的打包生成的固件也是不能启动。
> > > 自己编译的openwrt固件Target System选择的是QEMU ARM Virtual Machine,Subtarget是ARMv8
> > > 其他都没有改动。U盘没有问题，用其他固件可以正常启动。
> > 
> > 
> > 是连显示器看的嘛？确实看不到OpenWrt这个画面。不过用网线连接进后台正常
> 
> 是连的显示器，启动代码不再跑之后，按回车键不能退出。用网线也连不上n1,感觉是没启动完成。

昨天的确是有点问题，现在可以再试一下

nnssz, 2020-04-04:
> 哦，难怪切换很慢，150balbes的记忆
> 那个用armbian不错，做op真的不合适
> 不知道什么原因，flippy的没这个问题，但是没有xq7那个稳定

俄罗斯大神armbian20200205的内核是5.x，openwrt目前最新内核是4.x。会不会是内核不太适配的问题。tuanqing大佬可以考虑换成4.x的armbian底包试试。

hcym, 2020-04-04:
不一定是内核版本，f大佬的也是5.4，没有这个问题，刚刚用博主的生成op，运行正常，bbr正常，ipv6正常，无线我没有编译。还是切换开关节点的问题，cpu瞬间冲击2.0，然后慢慢恢复到零点

susanooxyz, 2020-04-04:
> > > > 用自己编译的openwrt打包的固件不能启动，用openwrt目录中自带的打包生成的固件也是不能启动。
> > > > 自己编译的openwrt固件Target System选择的是QEMU ARM Virtual Machine,Subtarget是ARMv8
> > > > 其他都没有改动。U盘没有问题，用其他固件可以正常启动。
> > > 
> > > 
> > > 是连显示器看的嘛？确实看不到OpenWrt这个画面。不过用网线连接进后台正常
> > 
> > 
> > 是连的显示器，启动代码不再跑之后，按回车键不能退出。用网线也连不上n1,感觉是没启动完成。
> 
> 昨天的确是有点问题，现在可以再试一下

还是不行。和昨天的一样。
我之前编译过其他固件，但是第一次编译n1的固件，不知道自己编译的openwrt有没有问题，所以想连上显示器和键盘先看一看。
现在不管是用我编译的openwrt还是你提供的openwrt，都不行，用网线也是连不上。
启动完后，就是图片中的样子

[image]

hcym, 2020-04-04:
哦，还有点小问题，生成的镜像文件没有解除权限，必须用sudo才能移动改名，还有一个boot空目录忘掉了，需要手动一下子

tuanqing, 2020-04-04:
> 哦，还有点小问题，生成的镜像文件没有解除权限，必须用sudo才能移动改名，还有一个boot空目录忘掉了，需要手动一下子

安排上了

tuanqing, 2020-04-04:
> 还是不行。和昨天的一样。
> 我之前编译过其他固件，但是第一次编译n1的固件，不知道自己编译的openwrt有没有问题，所以想连上显示器和键盘先看一看。
> 现在不管是用我编译的openwrt还是你提供的openwrt，都不行，用网线也是连不上。
> 启动完后，就是图片中的样子
>
> [image]

图片看不到，有啥问题去群里反馈吧，沟通方便一点

closed 2020-04-04 by susanooxyz