tuanqing/mknop #17: 运行出现如下错误：

liuhejun2020, 2020-04-10:
trojan@by:~/mknop$ sudo ./make
 i: firmware: 
 1. openwrt-bcm27xx-bcm2710-rpi-3-ext4-factory.img
 extract openwrt files... 
 extract armbian files... 
tar (child): /home/trojan/mknop/armbian/phicomm-n1/boot.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
tar (child): /home/trojan/mknop/armbian/phicomm-n1/modules.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
tar (child): /home/trojan/mknop/armbian/phicomm-n1/firmware.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now

 i: input the ROOTFS partition size, default 512m, do not less than 256m

 if you don't know what's the means, press enter to keep the default: 450

 make openwrt image... 
 format openwrt image... 
 copy files to image... 
cp: cannot stat '/home/trojan/mknop/tmp/boot/*': No such file or directory
 all done, enjoy!

liuhejun2020, 2020-04-11:
用你的脚本制作出现以上报错，然后刷机了，不启动

tuanqing, 2020-04-11:
> 用你的脚本制作出现以上报错，然后刷机了，不启动

boot,tar.gz、modules.tar.gz、firmware.tar.gz这三个文件删除了，等更新吧

hcym, 2020-04-11:
今天早上的版本完美，感谢大佬，感谢

closed 2020-04-30 by tuanqing