tuanqing/mknop #18: 出现以下问题，如何处理？

shenzt68, 2020-04-19:
mount: /home/szt/mknop/tmp/mount/boot: wrong fs type, bad option, bad superblock on /dev/loop0p1, missing codepage or helper program, or other error.
cp: error writing '/home/szt/mknop/tmp/mount/root/usr/lib/libbfd-2.27.so': No space left on device
cp: error writing '/home/szt/mknop/tmp/mount/root/usr/lib/libbind9.so.160.0.4': No space left on device
cp: error writing '/home/szt/mknop/tmp/mount/root/usr/lib/libblkid.so.1.1.0': No space left on device
cp: error writing '/home/szt/mknop/tmp/mount/root/usr/lib/libboost_chrono.so.1.71.0': No space left on device
cp: error writing '/home/szt/mknop/tmp/mount/root/usr/lib/libboost_date_time.so.1.71.0': No space left on device
cp: error writing '/home/szt/mknop/tmp/mount/root/usr/lib/libboost_program_options.so.1.71.0': No space left on device

tuanqing, 2020-04-19:
分区给小了

tuanqing, 2020-04-19:
不支持的img类型

shenzt68, 2020-04-19:
n1编译选项：
Target System (QEMU ARM Virtual Machine)  --->
Subtarget (ARMv8 multiplatform)  --->
Target Profile (Default)  --->
生成了：
config.seed                         
openwrt-armvirt-64-default-rootfs.tar.gz  
openwrt-armvirt-64-Image-initramfs  
openwrt-armvirt-64-root.squashfs.gz  
sha256sums
openwrt-armvirt-64-default.manifest  
openwrt-armvirt-64-Image                  
openwrt-armvirt-64-rootfs.cpio.gz   
packages
然后选择openwrt-armvirt-64-default-rootfs.tar.gz  放在operwrt里，执行make的。
请问哪步错了？

tuanqing, 2020-04-19:
没有问题，分区稍微给大一点看

closed 2020-04-30 by tuanqing