tuanqing/mknop #26: 在lxc中使用ubuntu 18.04模板创建的容器内生成N1固件报错

ZqinKing, 2020-05-29:
[ 5.4.20 ] extract openwrt files
 [ 5.4.20 ] extract armbian files
 [ 5.4.20 ] make openwrt image
 [ 5.4.20 ] format openwrt image
losetup: cannot find an unused loop device
 [ Error ] you used a lower version linux,
 please update the util-linux package or upgrade your system.

使用VM则没有问题。

ZqinKing, 2020-05-29:
ii  util-linux                    2.31.1-0.4ubuntu3.6                 amd64
提示的util-linux版本已升级至ubuntu18的最新版本

hcym, 2020-05-29:
虚拟化没用过，实体机没什么事，今天大佬构建版已经写入emmc，夏天过去再折腾啦，感谢cctx，呵呵😄

tuanqing, 2020-05-29:
> [ 5.4.20 ] extract openwrt files
> [ 5.4.20 ] extract armbian files
> [ 5.4.20 ] make openwrt image
> [ 5.4.20 ] format openwrt image
> losetup: cannot find an unused loop device
> [ Error ] you used a lower version linux,
> please update the util-linux package or upgrade your system.
> 
> 使用VM则没有问题。

内核不支持挂载loop设备

ZqinKing, 2020-05-29:
> > [ 5.4.20 ] extract openwrt files
> > [ 5.4.20 ] extract armbian files
> > [ 5.4.20 ] make openwrt image
> > [ 5.4.20 ] format openwrt image
> > losetup: cannot find an unused loop device
> > [ Error ] you used a lower version linux,
> > please update the util-linux package or upgrade your system.
> > 使用VM则没有问题。
> 
> 内核不支持挂载loop设备

嗯，搞清楚了是lxc默认没有loop导致的，需要在宿主机配置好了才行。

closed 2020-05-29 by ZqinKing