tuanqing/mknop #27: 使用win10子系统WSL制作镜像的问题

beenbin, 2020-05-29:
用的Actions云编译了N1的“ARMv8 multiplatform” 编译OP固件，clone仓库到wsl本地后，并且拷贝了openwrt-armvirt-64-default-rootfs.tar.gz 到openwrt文件夹后，开始制作后提示系统版本不是最新，可是我已经更换成国内阿里云源并且update和upgrade过了，还是这么提示，有高手解决么？
     PS:也尝试过橙子等网上的使用OpenWrt固件和Armbian镜像拼包制作可用于U盘启动的OpenWrt固件的教程，在挂载镜像的步骤就出错，无法挂载。本人纯小白，不知道是不是编译的OP固件没有.img镜像的原因，所以来试试这个云制作，还是出错了，求高手指导

input the the rootfs partition size, defaults to 512m, do not less than 256m                                                                                                              if you don't know what this means, press Enter to keep default:                                                                                                                           [ 512 ]                                                                                                                                                                                                                                                                                                                                                                             [ 5.4.20 ] extract openwrt files                                                                                                                                                          [ 5.4.20 ] extract armbian files                                                                                                                                                          [ 5.4.20 ] make openwrt image                                                                                                                                                             [ 5.4.20 ] format openwrt image                                                                                                                                                          Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.                                                                                                 Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.                                                                                                 Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.                                                                                                 losetup: cannot find an unused loop device                                                                                                                                                 [ Error ] you used a lower version linux,                                                                                                                                                 please update the util-linux package or upgrade your system.

beenbin, 2020-05-29:
/mknop$ sudo ./make                                                                                                                                                           Welcome to phicomm-n1 openwrt image tools!                                                                                                                                                firmware:                                                                                                                                                                                 (1) openwrt-armvirt-64-default-rootfs.tar.gz                                                                                                                                              [ openwrt-armvirt-64-default-rootfs.tar.gz ]                                                                                                                                                                                                                                                                                                                                        kernel:                                                                                                                                                                                   (1) 4.18.7                                                                                                                                                                                (2) 4.19.106                                                                                                                                                                              (3) 5.4.20                                                                                                                                                                                                                                                                                                                                                                          select the kernel above and press Enter to select the first one: 3                                                                                                                        [ 5.4.20 ]                                                                                                                                                                                                                                                                                                                                                                          input the the rootfs partition size, defaults to 512m, do not less than 256m                                                                                                              if you don't know what this means, press Enter to keep default:                                                                                                                           [ 512 ]                                                                                                                                                                                                                                                                                                                                                                             [ 5.4.20 ] extract openwrt files                                                                                                                                                          [ 5.4.20 ] extract armbian files                                                                                                                                                          [ 5.4.20 ] make openwrt image                                                                                                                                                             [ 5.4.20 ] format openwrt image                                                                                                                                                          Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.                                                                                                 Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.                                                                                                 Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.                                                                                                 losetup: cannot find an unused loop device                                                                                                                                                 [ Error ] you used a lower version linux,                                                                                                                                                 please update the util-linux package or upgrade your system.

tuanqing, 2020-05-29:
WSL-1 不支持挂载loop设备，WSL-2 ok

beenbin, 2020-05-29:
> WSL-1 不支持挂载loop设备，WSL-2 ok
哦哦哦，谢谢，我试试升级到2试试

closed 2020-05-29 by beenbin