tuanqing/mknop #31: 不同源码有的可以成功，有的不行。

qq892767935, 2020-07-07:
用Lean的源码编译的固件可以完美写入emmc，但用Lienol的源码编译的固件就不行，不知何故？ 好像提示分区失败。

tuanqing, 2020-07-08:
> 用Lean的源码编译的固件可以完美写入emmc，但用Lienol的源码编译的固件就不行，不知何故？ 好像提示分区失败。

parted这个包加了吗？

qq892767935, 2020-07-08:
.config - OpenWrt Configuration
 > Utilities > Disc ─────────────────────────────────────────────────────────────────────────────────────────────
  ┌────────────────────────────────────────────────── Disc ───────────────────────────────────────────────────┐
  │  Arrow keys navigate the menu.  <Enter> selects submenus ---> (or empty submenus ----).  Highlighted      │  
  │  letters are hotkeys.  Pressing <Y> includes, <N> excludes, <M> modularizes features.  Press <Esc><Esc>   │  
  │  to exit, <?> for Help, </> for Search.  Legend: [*] built-in  [ ] excluded  <M> module  < > module       │  
  │  capable                                                                                                  │  
  │ ┌───────────────────────────────────────────────────────────────────────────────────────────────────────┐ │  
  │ │                < > blkdiscard................................... discard sectors on a device          │ │  
  │ │                -*- blkid........................... locate and print block device attributes          │ │  
  │ │                < > blockdev.................. call block device ioctls from the command line          │ │  
  │ │                < > cfdisk........................ display or manipulate disk partition table          │ │  
  │ │                < > cgdisk........... GPT aware disk manipulation utility for interactive use          │ │  
  │ │                < > eject.............................................. eject removable media          │ │  
  │ │                <*> fdisk.................................... manipulate disk partition table          │ │  
  │ │                < > findfs................................ find a filesystem by label or UUID          │ │  
  │ │                < > fio.................................... I/O testing and benchmarking tool          │ │  
  │ │                < > fixparts....................... A utility to fix corrupted MBR partitions          │ │  
  │ │                < > gdisk............ GPT aware disk manipulation utility for interactive use          │ │  
  │ │                < > hd-idle..................... Another idle-daemon for attached hard drives          │ │  
  │ │                < > hdparm........................... Hard disk drive configuration utilitity          │ │  
  │ │                -*- lsblk................................................. list block devices          │ │  
  │ │                < > lvm2.................................... The Linux Logical Volume Manager          │ │  
  │ │                < > mdadm.......................... A tool for managing Soft RAID under Linux          │ │  
  │ │                -*- parted........................................... parted Partition editor          │ │  
  │ │                <*> partx-utils                                                                        │ │  
  │ │                < > sfdisk............................. partition table manipulator for Linux          │ │  
  │ │                < > sgdisk............. GPT aware disk manipulation utility for scripting use          │ │  
  │ │                < > wipefs.................................... wipe a signature from a device          │ │  
  │ │                                                                                                       │ │  
  │ │

tuanqing, 2020-07-08:
安装日志贴一下

qq892767935, 2020-07-08:
-----------------------------------------------------
root@OpenWrt:~# n1-install
 emmc: /dev/mmcblk1
 usb:  /dev/sda
 mount bootfs...
 copy bootable file...
mount: mounting /dev/sda1 on /boot failed: Invalid argument
cp: can't stat '/boot/*': No such file or directory
 edit uEnv.txt...
sed: /install/boot/uEnv.txt: No such file or directory
mv: can't rename '/install/boot/boot-emmc.scr': No such file or directory
 umount bootfs...
 mount rootfs...
 copy rootfs...
  --> copy bin...
  --> copy etc...
  --> copy lib...
  --> copy root...
  --> copy sbin...
  --> copy usr...
  --> copy www...
  --> link lib64...
  --> link var...
 edit fstab...
 sync...
 umount rootfs...
 copy openwrt os to emmc done!
root@OpenWrt:~#

qq892767935, 2020-07-08:
root@OpenWrt:~# n1-install
 emmc: /dev/mmcblk1
 usb:  /dev/sda
 mount bootfs...
 copy bootable file...
mount: mounting /dev/sda1 on /boot failed: Invalid argument
cp: can't stat '/boot/*': No such file or directory
 edit uEnv.txt...
sed: /install/boot/uEnv.txt: No such file or directory
mv: can't rename '/install/boot/boot-emmc.scr': No such file or directory
 umount bootfs...
 mount rootfs...
 copy rootfs...
  --> copy bin...
  --> copy etc...
  --> copy lib...
  --> copy root...
  --> copy sbin...
  --> copy usr...
  --> copy www...
  --> link lib64...
  --> link var...
 edit fstab...
 sync...
 umount rootfs...
 copy openwrt os to emmc done!
root@OpenWrt:~#

tuanqing, 2020-07-09:
同步最新的install-program试下

qq892767935, 2020-07-09:
每次编译都是下载最新源码，Lionel旧源码op19文件夹全删除，从头开始。难道是Lionel源码和lean的文件系统不一样？第一句错误是语法错误，但在lean的源码编译的固件却可以成功生成固件，并成功刷入emmc。Lionel源码也可以生成固件并成功用u盘启动openwrt,但写入emmc却提示错误。

tuanqing, 2020-07-09:
> 每次编译都是下载最新源码，Lionel旧源码op19文件夹全删除，从头开始。难道是Lionel源码和lean的文件系统不一样？第一句错误是语法错误，但在lean的源码编译的固件却可以成功生成固件，并成功刷入emmc。Lionel源码也可以生成固件并成功用u盘启动openwrt,但写入emmc却提示错误。

今天修正了，理论上没问题了，你可以试下。有问题再带截图反馈

qq892767935, 2020-07-09:
[image]

qq892767935, 2020-07-09:
[image]
[image]
[image]
![Uploading 截图录屏_dde-file-manager_20200709211412.png…](
[image]

大神帮忙看看   个人比较喜欢用lionel的源码。

qq892767935, 2020-07-09:
Lienol
打错字了

tuanqing, 2020-07-09:
同步下install-program

qq892767935, 2020-07-09:
[image]
问题依旧呢0.0

tuanqing, 2020-07-10:
现在呢

qq892767935, 2020-07-10:
[image]
重新拉去你的一键脚本源码，还是一样呢。

qq892767935, 2020-07-10:
[image]
重新拉取你的一键源码制作，还是一样呢。

qq892767935, 2020-07-10:
重新拉取你的一键源码制作，还是一样呢。
[image]

tuanqing, 2020-07-10:
把你的固件传过来看看

tuanqing, 2020-07-10:
把你的固件发过来看一下，谢谢

qq892767935, 2020-07-10:
https://yangmantang.lanzous.com/iZFxuehijha

https://cloud.189.cn/t/iMraIneieyI3

qq892767935, 2020-07-10:
限制10M没法上传，只能放网盘了 。麻烦大神了。

tuanqing, 2020-07-10:
> 限制10M没法上传，只能放网盘了 。麻烦大神了。

https://cloud.189.cn/t/yiMV3aVj6Bv2
可以了

qq892767935, 2020-07-10:
先谢谢大神，等下吃完饭试下，顺便问下，大神问题何故？

tuanqing, 2020-07-10:
> 先谢谢大神，等下吃完饭试下，顺便问下，大神问题何故？

我的锅，判断条件不严谨

closed 2020-07-10 by tuanqing

qq892767935, 2020-07-10:
谢谢大佬

tuanqing, 2020-07-11:
> 谢谢大佬

应该🆗了吧？下次使用最新的install-program就行了