tuanqing/mknop #32: 编译提示错误dd: invalid number: ''

cyqco7612, 2020-07-07:
用的lean的源码，打包时提示错误如下，
dd if=/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/root.squashfs of=/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/root.squashfs.new bs= conv=sync
dd: invalid number: ''
Makefile:34: recipe for target 'install-images' failed
make[5]: *** [install-images] Error 1
make[5]: Leaving directory '/home/ubuntu/lede/target/linux/armvirt/image'
Makefile:20: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/ubuntu/lede/target/linux/armvirt'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/target/linux'
time: target/linux/install#409.11#58.29#414.83
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/ubuntu/lede'
target/Makefile:19: recipe for target '/home/ubuntu/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install' failed
make[1]: *** [/home/ubuntu/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
/home/ubuntu/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
但是已经打包的部分可以用。不知道这个错误有什么影响。

tuanqing, 2020-07-08:
> 用的lean的源码，打包时提示错误如下，
> dd if=/home/ubuntu/lede/build_dir/target-aarch64_generic_musl/linux-armvirt_64/root.squashfs [...]

不影响，编译时取消squashfs就行了

closed 2020-07-10 by tuanqing

cyqco7612, 2020-07-31:
谢谢