tuanqing/mknop #37: [ Error ] mount /dev/loop8p2 failed!

yehoshualee, 2020-08-10:
Unbuntu 18.04版本系统,有下面的错误提示:
bdder@ubuntu:~/openwrt/mknop$ sudo ./make 
[sudo] bdder 的密码： 
 Welcome to phicomm-n1 openwrt image tools!

 firmware: 
 (1) openwrt-armvirt-64-rootfs-ext4.img
 [ openwrt-armvirt-64-rootfs-ext4.img ]

 kernel: 
 (1) 4.19.134
 (2) 5.4.50

 select kernel above, and press Enter to select the first one: 2
 [ 5.4.50 ]

 input the rootfs partition size, defaults to 512m, do not less than 256m
 if you don't know what this means, press Enter to keep default: 
 [ 512 ]

 [ 5.4.50 ] extract openwrt files 
mount: /home/bdder/openwrt/mknop/tmp/mount: 特殊设备 /dev/loop8p2 不存在.
 [ Error ] mount /dev/loop8p2 failed!

hcym, 2020-08-10:
别用ext4的，不知道从那个版本开始不行了，以前两种格式的都可以

necoic, 2020-08-10:
我18.04没问题

hcym, 2020-08-10:
哦，这样啊，呵呵😄，无所谓啦

hcym, 2020-08-10:
舔什么乱，楼主是1804，我是都换2004了，等大佬看看吧🤣

yehoshualee, 2020-08-10:
> 舔什么乱，楼主是1804，我是都换2004了，等大佬看看吧🤣

我用2004也不行,如果你们没有类似问题也可能是我自己没搞好

yehoshualee, 2020-08-10:
换成其他格式的好了

closed 2020-08-10 by yehoshualee