tuanqing/mknop #41: 写入U盘后启动失败

bin20088, 2020-08-29:
编译完op后用这个源码制作出img镜像，写入U盘后启动失败，如图。另外我想问一下，如果是电视盒子状态，还需要把armbian底包刷入emmc吗？
[image]

hcym, 2020-08-29:
U盘启动使用都正常了再写emmc，其实无所谓的，最近各种app都要变动，写来写去的麻烦，我基本上都是用U盘系统

bin20088, 2020-08-29:
经过反复测试，同样源码同样的编译环境，固件里集成插件少的一切正常，多的就这个结果。编译用的这个架构。CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2710=y
CONFIG_TARGET_bcm27xx_bcm2710_DEVICE_rpi-3=y
[image]

Margular, 2020-08-30:
同样，启动失败，卡在N1盒子启动的第一个界面就不动了我是，然后拔掉U盘再重启就正常进入原系统了

hcym, 2020-08-31:
可能有个体差异，我的n1杂食动物，老的512m和1G的U盘也能用，就是编译打包时注意到选择最小就可以，大佬的最小可以400M，运行完美😹

goer99, 2020-09-04:
也遇到过与楼主类似的USB启动失败情况，必然发生kernel panic，百思不得其解。

写盘时使用的是一直在使用的USBWriter（只有176K），后来尝试把写盘软件换成作者所推荐的 Etcher （100多M），就启动正常了。奇怪的是，flippy编译的N1固件，使用USBWriter写盘就没问题。

不知道楼主写盘时是否用的 Etcher，如果不是，可以换成 Etcher试试。

> 编译完op后用这个源码制作出img镜像，写入U盘后启动失败，如图。另外我想问一下，如果是电视盒子状态，还需要把armbian底包刷入emmc吗？
> [image]

bin000000, 2020-09-04:
确实是你说的这样，搞定了。谢谢

> 也遇到过与楼主类似的USB启动失败情况，必然发生kernel panic，百思不得其解。
> 
> 写盘时使用的是一直在使用的USBWriter（只有176K），后来尝试把写盘软件换成作者 tuanqing 所推荐的 Etcher （100多M），就启动正常了。奇怪的是，flippy编译的N1固件，使用USBWriter写盘就没问题。
> 
> 不知道楼主写盘时是否用的 Etcher，如果不是，可以换成 Etcher试试。
> 
> > 编译完op后用这个源码制作出img镜像，写入U盘后启动失败，如图。另外我想问一下，如果是电视盒子状态，还需要把armbian底包刷入emmc吗？
> > [image]

closed 2020-09-04 by bin20088