tuanqing/mknop #49: 想咨询下关于内核版本的问题

ylqjgm, 2020-10-02:
对于工具中的内核版本和OP编译的内核版本之间的关系不太了解，想请问下：

比如我使用Lienol源码编译的，内核版本是4.19，而工具中的内核版本是44+o的，是5.4.66
请问下，其中5.4.66这个内核是不是属于底包的内核，也就是armbian的内核，而4.19是我OP的内核，两个互相并不干扰，我写入到N1后，在OP后台看到的内核版本依然是4.19，不是5.4，请问是这个意思么？
如果我编译的是4.19，那请问下我是否应该使用对应的27o的4.19内核来打包，这样稳定呢？还是随意都稳定？

tuanqing, 2020-10-02:
> 对于工具中的内核版本和OP编译的内核版本之间的关系不太了解，想请问下：
> 
> 比如我使用Lienol源码编译的，内核版本是4.19，而工具中的内核版本是44+o的，是5.4.66
> 请问下，其中5.4.66这个内核是不是属于底包的内核，也就是armbian的内核，而4.19是我OP的内核，两个互相并不干扰，我写入到N1后，在OP后台看到的内核版本依然是4.19，不是5.4，请问是这个意思么？
> 如果我编译的是4.19，那请问下我是否应该使用对应的27o的4.19内核来打包，这样稳定呢？还是随意都稳定？

你好，这个相当于完全摒弃了 openwrt 内核，使用 armbian 内核，也就是说启动之后看到的内核版本是 5.4 而不是 4.19。
至于稳定性，它受制于 openwrt 固件的稳定性和内核本身的稳定性，因此具体请以自身测试为准。

ylqjgm, 2020-10-03:
> 你好，这个相当于完全摒弃了 openwrt 内核，使用 armbian 内核，也就是说启动之后看到的内核版本是 5.4 而不是 4.19。
> 至于稳定性，它受制于 openwrt 固件的稳定性和内核本身的稳定性，因此具体请以自身测试为准。

感谢回答，意思就是说，无论我openwrt编译的内核版本是什么，到最后我都会被armbian的内核替换掉，直接更换为armbian内核，是这样么？

tuanqing, 2020-10-03:
> 感谢回答，意思就是说，无论我openwrt编译的内核版本是什么，到最后我都会被armbian的内核替换掉，直接更换为armbian内核，是这样么？

是的

closed 2020-10-03 by tuanqing