tuanqing/mknop #52: feeds缺少大量依赖

liwenjie119, 2020-10-05:
拉取代码，稍微修改了一下，然后发现编译出错，在编译openwrt阶段就出问题，于是拉代码到本地运行config分析，发现缺少大量依赖，不知道原版是否同样问题：
`WARNING: Makefile 'package/feeds/packages/audit/Makefile' has a dependency on 'libaudit', which does not exist
WARNING: Makefile 'package/feeds/packages/audit/Makefile' has a dependency on 'libaudit', which does not exist
WARNING: Makefile 'package/feeds/packages/audit/Makefile' has a dependency on 'libaudit', which does not exist
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/feeds/packages/libcgroup/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/libs/libselinux/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/feeds/packages/libzip/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/feeds/packages/nfdump/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/feeds/packages/python3-libselinux/Makefile' has a dependency on 'musl-fts', which does not exist
WARNING: Makefile 'package/feeds/packages/python3-libsemanage/Makefile' has a dependency on 'libsemanage', which does not exist
WARNING: Makefile 'package/feeds/packages/semodule-utils/Makefile' has a build dependency on 'libsemanage/host', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.`

tuanqing, 2020-10-05:
请使用 lean packages

tuanqing, 2020-10-05:
> 拉取代码，稍微修改了一下，然后发现编译出错，在编译openwrt阶段就出问题，于是拉代码到本地运行config分析，发现缺少大量依赖，不知道原版是否同样问题：
> `WARNING: Makefile 'package/feeds/packages/audit/Makefile' has a dependency on 'libaudit', which does not exist [...]

你好，该问题已在 [d6c8ce6e058f17fd1fbcf1246212186c4770a789](https://github.com/tuanqing/mknop/commit/d6c8ce6e058f17fd1fbcf1246212186c4770a789) 中解决

liwenjie119, 2020-10-05:
> 请使用 lean packages

OK，我更新一下

closed 2020-10-27 by tuanqing