tuanqing/mknop #53: 打包时报错

WROIATE, 2020-10-08:
lsblk: failed to access sysfs directory: /sys/dev/block: No such file or directory

e => error when losetup out/phicomm-n1/20.10.08-114830-phicomm-n1-5.4.69-aarch64-openwrt-firmware.img
lsblk: failed to access sysfs directory: /sys/dev/block: No such file or directory

但是看输出目录也生成了固件，不知道有没有影响

tuanqing, 2020-10-08:
> lsblk: failed to access sysfs directory: /sys/dev/block: No such file or directory
> 
> e => error when losetup out/phicomm-n1/20.10.08-114830-phicomm-n1-5.4.69-aarch64-openwrt-firmware.img
> lsblk: failed to access sysfs directory: /sys/dev/block: No such file or directory
> 
> 但是看输出目录也生成了固件，不知道有没有影响

麻烦贴完整日志

WROIATE, 2020-10-08:
$ sudo ./gen_openwrt -d
Welcome to aarch64 openwrt firmware generator!

lsblk: failed to access sysfs directory: /sys/dev/block: No such file or directory
device          =>  beikeyun, l1pro, phicomm-n1, s905x3, vplus
firmware        =>  openwrt-armvirt-64-default-rootfs.tar.gz
kernel          =>  5.4.69
rootsize        =>  512

unpack openwrt
unpack armbian  =>  beikeyun @ 5.4.69
unpack armbian  =>  l1pro @ 5.4.69
unpack armbian  =>  phicomm-n1 @ 5.4.69
unpack armbian  =>  s905x3 @ 5.4.69
unpack armbian  =>  vplus @ 5.4.69
make firmware   =>  vplus @ 5.4.69
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
make firmware   =>  beikeyun @ 5.4.69
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
make firmware   =>  phicomm-n1 @ 5.4.69
make firmware   =>  s905x3 @ 5.4.69
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
make firmware   =>  l1pro @ 5.4.69
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
e => error when losetup out/vplus/20.10.08-114610-vplus-5.4.69-aarch64-openwrt-firmware.img
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
e => error when losetup out/beikeyun/20.10.08-114610-beikeyun-5.4.69-aarch64-openwrt-firmware.img
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
e => error when losetup out/phicomm-n1/20.10.08-114611-phicomm-n1-5.4.69-aarch64-openwrt-firmware.img
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
e => error when losetup out/s905x3/20.10.08-114611-s905x3-5.4.69-aarch64-openwrt-firmware.img
Kernel not configured for semaphores (System V IPC). Not using udev synchronisation code.
losetup: cannot find an unused loop device
e => error when losetup out/l1pro/20.10.08-114611-l1pro-5.4.69-aarch64-openwrt-firmware.img
lsblk: failed to access sysfs directory: /sys/dev/block: No such file or directory

tuanqing, 2020-10-08:
WSL？

WROIATE, 2020-10-08:
> WSL？

是的

tuanqing, 2020-10-08:
> > WSL？
> 
> 是的

升级 WSL2 吧，如果还是不行就是不支持

WROIATE, 2020-10-08:
好的，了解了，谢谢作者

closed 2020-10-08 by WROIATE