tuanqing/mknop #56: n1-update失败

lexin8, 2020-10-20:
执行 n1-install 安装到 emmc，将固件上传到 /tmp/upgrade( xxx.img )，之后执行 n1-update

root@OpenWrt:~# n1-update
Error: you have never installed a OS to emmc, please boot with usb and run n1-install to install at first!

tuanqing, 2020-10-20:
> 执行 n1-install 安装到 emmc，将固件上传到 /tmp/upgrade( xxx.img )，之后执行 n1-update
> 
> root@OpenWrt:~# n1-update
> Error: you have never installed a OS to emmc, please boot with usb and run n1-install to install at first!

执行 `lsblk ` 看下

lexin8, 2020-10-20:
> > 执行 n1-install 安装到 emmc，将固件上传到 /tmp/upgrade( xxx.img )，之后执行 n1-update
> > root@OpenWrt:~# n1-update
> > Error: you have never installed a OS to emmc, please boot with usb and run n1-install to install at first!
> 
> 执行 `lsblk ` 看下

[image]

tuanqing, 2020-10-20:
> > > 执行 n1-install 安装到 emmc，将固件上传到 /tmp/upgrade( xxx.img )，之后执行 n1-update
> > > root@OpenWrt:~# n1-update
> > > Error: you have never installed a OS to emmc, please boot with usb and run n1-install to install at first!
> > 
> > 
> > 执行 `lsblk ` 看下
> 
> [image]

先 `losetup -d /dev/loop0` ，然后再 `n1-update` 试试。
还不行的话就 `blkid` 然后贴张图过来看看

lexin8, 2020-10-21:
> 先 `losetup -d /dev/loop0` ，然后再 `n1-update` 试试。
> 还不行的话就 `blkid` 然后贴张图过来看看

[image]

tuanqing, 2020-10-21:
你好，请更新下 `install-program` ，应该是没问题了

lexin8, 2020-10-21:
> 你好，请更新下 `install-program` ，应该是没问题了

重新编译新固件吗

closed 2020-10-22 by lexin8