tuanqing/mknop #61: 执行n1-install总是提示空间不足，刷入EMMC失败

keeno1982, 2020-11-02:
root@OpenWrt:/# n1-install
emmc: /dev/mmcblk1
usb:  /dev/sda

backup u-boot...
create mbr and partition...
restore u-boot...
format boot partiton...
format root partiton...
e2fsck 1.45.6 (20-Mar-2020)
format data partiton...
e2fsck 1.45.6 (20-Mar-2020)
copy bootable file...
edit uEnv...
copy rootfs...
 --> copy bin...
 --> copy etc...
 --> copy lib...
tar: write error: No space left on device
tar: write error: Broken pipe
 --> copy root...
tar: can't make dir root: No space left on device
tar: can't create directory 'root': No space left on device
tar: can't open 'root/.wgetrc': No such file or directory
 --> copy sbin...
tar: can't make dir sbin: No space left on device
tar: can't create directory 'sbin': No space left on device
tar: can't open 'sbin/upgraded': No such file or directory
tar: write error: Broken pipe
 --> copy usr...
tar: can't make dir usr: No space left on device
tar: can't create directory 'usr': No space left on device
tar: can't make dir usr/sbin: No such file or directory
tar: can't create directory 'usr/': No space left on device
tar: can't create symlink 'usr/' to 'xtables-legacy-multi': No such file or directory
tar: write error: Broken pipe
 --> copy www...
tar: can't make dir www: No space left on device
tar: can't create directory 'www': No space left on device
tar: can't make dir www/cgi-bin: No such file or directory
tar: can't create directory 'www/': No space left on device
tar: can't open 'www/': Is a directory
tar: write error: Broken pipe
 --> link lib64...
 --> link var...
cp: can't stat 'etc/config/fstab': No such file or directory
cp: can't stat 'etc/config/fstab.bak': No such file or directory
edit fstab...
sed: etc/config/fstab: No such file or directory
sed: etc/config/fstab: No such file or directory
sed: etc/config/fstab: No such file or directory
/usr/bin/n1-install: cd: line 148: can't cd to lib/firmware/brcm: No such file or directory
sed: brcmfmac43455-sdio.phicomm,n1.txt: No such file or directory
sync...
copy openwrt os to emmc done!
root@OpenWrt:/#

tuanqing, 2020-11-03:
生成固件时，`rootfs` 分区大小设置大一些

keeno1982, 2020-11-03:
> 
> 
> 生成固件时，`rootfs` 分区大小设置大一些

貌似不是这个原因？我设成1G也是报错，难道要设成更大一些？

tuanqing, 2020-11-03:
> > 生成固件时，`rootfs` 分区大小设置大一些
> 
> 貌似不是这个原因？我设成1G也是报错，难道要设成更大一些？

我确认下，

keeno1982, 2020-11-03:
> 
> 
> > > 生成固件时，`rootfs` 分区大小设置大一些
> > 
> > 
> > 貌似不是这个原因？我设成1G也是报错，难道要设成更大一些？
> 
> 我确认下，

好的，谢谢大佬。有个疑问，我编译固件的时候设置rootfs为600M，那在生成N1固件时出现提示那边应该填多少呢？大于等于还是小于600？

tuanqing, 2020-11-03:
> > > > 生成固件时，`rootfs` 分区大小设置大一些
> > > 
> > > 
> > > 貌似不是这个原因？我设成1G也是报错，难道要设成更大一些？
> > 
> > 
> > 我确认下，
> 
> 好的，谢谢大佬。有个疑问，我编译固件的时候设置rootfs为600M，那在生成N1固件时出现提示那边应该填多少呢？大于等于还是小于600？

编译时无所谓，能装得下就行。生成时要大于 600M，因为生成时还需要拷贝 `modules` `firmware`  等内容。建议不要设置得太小，以免出现其他问题。

keeno1982, 2020-11-03:
> 
> 
> > > > > 生成固件时，`rootfs` 分区大小设置大一些
> > > > 
> > > > 
> > > > 貌似不是这个原因？我设成1G也是报错，难道要设成更大一些？
> > > 
> > > 
> > > 我确认下，
> > 
> > 
> > 好的，谢谢大佬。有个疑问，我编译固件的时候设置rootfs为600M，那在生成N1固件时出现提示那边应该填多少呢？大于等于还是小于600？
> 
> 编译时无所谓，能装得下就行。生成时要大于 600M，因为生成时还需要拷贝 `modules` `firmware` 等内容。建议不要设置得太小，以免出现其他问题。

有找到原因了吗？
刚才我设置成2G，也是失败。直接下载你编译好的也会失败，但是刷别人的固件是正常的，刷coreelec也没问题。

tuanqing, 2020-11-03:
你好，请手动修改下 `n1-install`
[参考这里](https://github.com/tuanqing/install-program/commit/b69d01761c3df608906201a648fac4cb99ca964b)
更新 `install-program` 再编译也行。

keeno1982, 2020-11-03:
@

> 
> 
> 你好，请手动修改下 `n1-install`
> [参考这里](https://github.com/tuanqing/install-program/commit/b69d01761c3df608906201a648fac4cb99ca964b)
> 更新 `install-program` 再编译也行。

太感谢了！完美！

closed 2020-11-05 by tuanqing