tuanqing/mknop #63: 使用n1-update变砖

ZqinKing, 2020-11-05:
上一次也是n1-update之后设备变砖。昨晚救砖成功。今天不信邪再次使用n1-update。问题成功复现。又变砖了。

tuanqing, 2020-11-05:
> 上一次也是n1-update之后设备变砖。昨晚救砖成功。今天不信邪再次使用n1-update。问题成功复现。又变砖了。

是不是提前断电了？

tuanqing, 2020-11-05:
你好，经过测试并未发现你所说的情况。
请确认步骤是否正确，如无误，请提供 `n1-update` 的运行日志。

ZqinKing, 2020-11-06:
> 你好，经过测试并未发现你所说的情况。
> 请确认步骤是否正确，如无误，请提供 `n1-update` 的运行日志。

这个日志在哪儿看？我是自己lean编译的openwrt做的 固件。

ZqinKing, 2020-11-06:
[http://kod.zqinking.tk:8088/index.php?share/file&user=1&sid=2aYwfT9K](url)固件用的是这个，昨晚救砖回来后用n1-install从U盘刷到emmc没有问题。之前同样这个固件在现有的系统中使用n1-update就会变砖。

> 你好，经过测试并未发现你所说的情况。
> 请确认步骤是否正确，如无误，请提供 `n1-update` 的运行日志。

[http://kod.zqinking.tk:8088/index.php?share/file&user=1&sid=2aYwfT9K](url)固件用的是这个，昨晚救砖回来后用n1-install从U盘刷到emmc没有问题。之前同样这个固件在现有的系统中使用n1-update就会变砖。

ZqinKing, 2020-11-06:
> 你好，经过测试并未发现你所说的情况。
> 请确认步骤是否正确，如无误，请提供 `n1-update` 的运行日志。

[http://kod.zqinking.tk:8088/index.php?share/file&user=1&sid=2aYwfT9K](url)固件用的是这个，昨晚救砖回来后用n1-install从U盘刷到emmc没有问题。之前同样这个固件在现有的系统中使用n1-update就会变砖。

tuanqing, 2020-11-06:
`update` 过程中有没有报错之类的，请提供运行截图好确认问题。

ZqinKing, 2020-11-09:
> `update` 过程中有没有报错之类的，请提供运行截图好确认问题。

没有报错。能到重启那儿

ihuaer, 2020-11-26:
> > `update` 过程中有没有报错之类的，请提供运行截图好确认问题。
> 
> 没有报错。能到重启那儿

其实没变砖，升级固件后没有自动重启，手动插拔电源就可以了，我也是这个情况

HOT-POT1, 2020-11-28:
copy bootable file...
edit uEnv...
mount: mounting /dev/mmcblk1p2 on /install/root failed: No error information
copy rootfs...
 --> copy bin...
 --> copy etc...
 --> copy lib...
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/mediatek': No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/rtl_bt': No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/ap6210': No space left on device
cp: can't create directory '/install/root/lib/firmware/xr819': No space left on device
cp: can't create directory '/install/root/lib/firmware/RTL8192SU': No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/rtl_nic': No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/edid': No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/meson': No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/imx': No space left on device
cp: can't create directory '/install/root/lib/firmware/rtlbt': No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/sdma': No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: can't create directory '/install/root/lib/firmware/vpu': No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
cp: write error: No space left on device
上面的报错是在我update之后，又重新install报的，update时没有报任何错，只是重启之后n1就没反应了，也就是和issue一样，解决办法是刷写f大的镜像（可能会报错），然后再重新install自己打包的固件就恢复了

ZqinKing, 2020-12-03:
> `update` 过程中有没有报错之类的，请提供运行截图好确认问题。

start upgrade from phicomm-n1-5.4.77-aarch64-openwrt-firmware.img...
losetup: /tmp/upgrade/phicomm-n1-5.4.77-aarch64-openwrt-firmware.img: Warning: file does not fit into a 512-byte sector; the end of the file will be ignored.
mount: mounting /dev/loop0p2 on /mnt/loop0p2 failed: Invalid argument
cp: can't stat '/mnt/loop0p2/etc/config/fstab': No such file or directory

do you want to save configuration? [Y/n] ^C

ZqinKing, 2020-12-03:
已解决，自行更改了最小包大小缩小到256后导致生成的包有问题。

closed 2020-12-03 by ZqinKing