tuanqing/mknop #76: 出错：设备上没有空间

JuneOx, 2021-02-06:
unpack openwrt
unpack armbian  =>  phicomm-n1 @ 5.4.86
make firmware   =>  phicomm-n1 @ 5.4.86
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/ajv/dist': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/ajv/lib': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/ajv/scripts': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/asn1': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/assert-plus': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/asynckit': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/aws-sign2': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/aws4': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/bcrypt-pbkdf': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/caseless': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/combined-stream': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/core-util-is': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/dashdash': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/delayed-stream': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/ecc-jsbn': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/extend': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/extsprintf': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/fast-deep-equal': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/fast-json-stable-stringify': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/forever-agent': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/form-data': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/getpass': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/har-schema': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/har-validator': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/http-signature': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/is-typedarray': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/isstream': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/jsbn': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/json-schema': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/json-schema-traverse': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/json-stringify-safe': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/jsprim': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/mime-db': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/mime-types': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/oauth-sign': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/performance-now': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/psl': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/punycode': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/qs': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/safe-buffer': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/safer-buffer': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/sshpk': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/tough-cookie': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/tunnel-agent': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/tweetnacl': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/uri-js': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/uuid': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/node/request/node_modules/verror': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/opkg': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/pppd': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/python3.9': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/rpcd': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/ruby': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/samba': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/sasl2': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/sqm': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/squid': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/tc': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/lib/zsh': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/libexec': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/sbin': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/usr/share': 设备上没有空间
cp: 无法创建目录 './tmp/mount/loop4p2/www': 设备上没有空间
rm: 无法删除 './tmp/mount/loop2': 设备或资源忙
rm: 无法删除 './tmp/mount/loop0': 设备或资源忙
rm: 无法删除 './tmp/mount/loop1': 设备或资源忙

tuanqing, 2021-02-06:
空间不足就增加 `rootfs` 分区大小

cattypapa, 2021-05-23:
把rootfs控件设置为1024，如果还是提示这个，就继续加大数值，就Ok了