tuanqing/mknop #84: -e 命令无法正确从openwrt的包中提取出kernel

lymzzyh, 2021-09-15:
[image]

如上图所示

tuanqing, 2021-09-16:
这个功能是从 `armbian ` 或其它经过打包的 `openwrt` 固件中提取内核

lymzzyh, 2021-09-16:
如果我想要使用openwrt的内核应该怎么做？毕竟有些内核模块是需要跟随内核一起编译的

tuanqing, 2021-09-17:
> 如果我想要使用openwrt的内核应该怎么做？毕竟有些内核模块是需要跟随内核一起编译的

不行，打包的方式即不使用 `openwrt kernel`
如果感兴趣，可以尝试 [openwrt-target-amlogic](https://github.com/chencaidy/openwrt-target-amlogic)
将此项目 `clone` 到 `target/linux` ，然后在 `make menuconfig` 中配置如下：
`Target System (Amlogic Meson)  --->
Subtarget (Meson GX/AXG/G12A boards)  --->
Target Profile (Phicomm N1)  --->`

tuanqing, 2021-09-17:
> 如果我想要使用openwrt的内核应该怎么做？毕竟有些内核模块是需要跟随内核一起编译的

不行，打包的方式即不使用 `openwrt kernel`
如果感兴趣，可以尝试 [openwrt-target-amlogic](https://github.com/chencaidy/openwrt-target-amlogic)
将此项目 `clone` 到 `target/linux` ，然后在 `make menuconfig` 中配置如下：
```
Target System (Amlogic Meson) --->  
Subtarget (Meson GX/AXG/G12A boards) --->  
Target Profile (Phicomm N1) --->
```

lymzzyh, 2021-09-17:
非常感谢你的耐心解答。

closed 2021-09-17 by lymzzyh